107-systems/107-Arduino-MCP2515 #10: Library not compiling on Windows

ChipLeo, 2020-04-25:
Path not found for Windows

per1234, 2020-04-25:
Hi @ChipLeo. It will be helpful if you provide some additional information:

When you encounter an error you'll see a button on the right side of the orange bar in the Arduino IDE "Copy error messages" (or the icon that looks like two pieces of paper at the top right corner of the black console window in the Arduino Web Editor). Click that button.

[Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error in a message here using [code fencing](https://guides.github.com/features/mastering-markdown/).

ChipLeo, 2020-04-25:
`In file included from D:\Leo\Documents\Arduino\libraries\107-Arduino-MCP2515\src/ArduinoMCP2515.h:16:0,

Несколько библиотек найдено для "ArduinoMCP2515.h"
                 from D:\Leo\Documents\Arduino\sketch_apr25a\sketch_apr25a.ino:2:

 Используется: D:\Leo\Documents\Arduino\libraries\107-Arduino-MCP2515
D:\Leo\Documents\Arduino\libraries\107-Arduino-MCP2515\src/MCP2515/MCP2515_Io.h:17:10: fatal error: functional: No such file or directory`

but library exist in 
D:\Leo\Documents\Arduino\libraries\107-Arduino-MCP2515

per1234, 2020-04-25:
You didn't post the important information of which board you are compiling for. I'm guessing it's one of the AVR boards.

ChipLeo, 2020-04-25:
Arduino: 1.8.12 (Windows 8.1), Плата:"Arduino Uno"

In file included from D:\Leo\Documents\Arduino\libraries\107-Arduino-MCP2515\src/ArduinoMCP2515.h:16:0,

                 from D:\Leo\Documents\Arduino\sketch_apr25a\sketch_apr25a.ino:2:

D:\Leo\Documents\Arduino\libraries\107-Arduino-MCP2515\src/MCP2515/MCP2515_Io.h:17:10: fatal error: functional: No such file or directory

 #include <functional>

          ^~~~~~~~~~~~

compilation terminated.

exit status 1
Ошибка компиляции для платы Arduino Uno.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

closed 2020-04-27 by aentinger

aentinger, 2020-04-27:
@ChipLeo you might want to take a look at https://github.com/autowp/arduino-mcp2515 ... it is an excellent library and should be available for AVR.

PR #11 (2020-04-27): Restrict list of supported architectures to samd
Since `107-Arduino-MCP2515` uses advanced C++ features such as `<functional>` it can not be compiled for AVR boards. Therefore it is a good idea to specify the supported architectures which is currently `samd` only.

This PR fixes #10 since it gives users a warning when compiling for a different architecture then samd.