107-systems/107-Arduino-MCP2515 #12: Transition this library from Travis to GitHub Action

aentinger, 2020-07-08:
Since the new GitHub Actions CI build system obviously is the way to go I'd like to transition this repository to the new system. @per1234 can I ask you to do that?

per1234, 2020-07-08:
No problem.

Some questions:
- Do you want size deltas reports?
- Do you want size trends reports?
- If yes to either of the above, which example should be the "size report sketch"?

aentinger, 2020-07-09:
Hi :wave:

>Do you want size deltas reports?

Yes.

>Do you want size trends reports?

I'd like to wait if/when the history over master feature is available, so no for now.

>If yes to either of the above, which example should be the "size report sketch"?

Please use [this sketch](https://github.com/107-systems/107-Arduino-MCP2515/blob/master/examples/MCP2515-Loopback/MCP2515-Loopback.ino).

Thank you 🚀

closed 2020-07-09 by aentinger