107-systems/107-Arduino-MCP2515 #25: use 29-bit CAN-IDs

generationmake, 2020-12-20:
at the moment the library uses 11-bit CAN-IDs but UAVCAN uses extended 29-bit CAN-IDs.
See this forum post: https://forum.uavcan.org/t/receiving-heartbeat-message-with-pyuavcan-cli/1035/4?u=generationmake

aentinger, 2020-12-20:
@pavel-kirienko You've got it right! Thank you and @generationmake for opening this issue, the fix is contained within #25.

generationmake, 2020-12-20:
@aentinger 
Seems to be OK now.
Output from candump:
```
 (1608480559.168992)  can0  RX - -  107D550D   [8]  3D 00 00 00 00 00 00 FD   '=.......'
 (1608480560.170775)  can0  RX - -  107D550D   [8]  3E 00 00 00 00 00 00 FE   '>.......'
 (1608480561.172638)  can0  RX - -  107D550D   [8]  3F 00 00 00 00 00 00 E0   '?.......'
```

Output from pyuavcan:
```
---
7509:
  uptime: 55
  health:
    value: 0
  mode:
    value: 0
  vendor_specific_status_code: 0

---
7509:
  uptime: 56
  health:
    value: 0
  mode:
    value: 0
  vendor_specific_status_code: 0
```

closed 2020-12-20 by generationmake

aentinger, 2020-12-20:
Juhu 🚀

PR #26 (2020-12-20): Bugfix: UAVCAN uses 29-bit CAN identifiers
Therefore we always need …to set the bit marking an extended frame when dealing with libcanard frames.

This fixes #25.