107-systems/107-Arduino-MCP2515 #36: Portenta with MRK CAN shield

guilleherreraf, 2021-02-23:
Hello! I'm trying to receive CAN frames with the Portenta and the [MKR CAN shield](https://store.arduino.cc/arduino-mkr-can-shield) using this library. However, I'm not able to receive anything. I'm trying the CAN sniffer example. The code compiles, but I do not see any data in the serial terminal. I visualize the CAN frame in a oscilloscope and it seems to be fine. Any idea why it might not be working?

aentinger, 2021-02-24:
Hi @guilleherreraf :wave: Probably you've got the wrong chip select pin or wrong SPI since the examples are written for and tested on the SAMD platform. Find and verify the right pinout by using [MCP2515-Loopback.ino](https://github.com/107-systems/107-Arduino-MCP2515/blob/master/examples/MCP2515-Loopback/MCP2515-Loopback.ino).

guilleherreraf, 2021-02-24:
Hello @aentinger. Thanks for your quick answer. I do not think that is the issue because I'm using the MKR CAN shield on the Portenta, which uses the CS pin 3 and the INT pin 7. Furthermore, the SPI pins are exactly the same, as the Portenta is compatible with the MKR boards. 
[image]
What I find strange is that the Portenta seems to crash, and the red LED blinks 4 times fast and 4 times slow. I have tried to find out what this means, but have not found any clue. At the beginning I thought it was a power source problem, but I also tried powering the MKR from the Vin pin and I get the same behavior.

aentinger, 2021-02-24:
Hi :wave: The LED pattern is the mbed core crashing. Can you attach a 3V3 TTL to USB converter to [UART1](https://content.arduino.cc/assets/Pinout-PortentaH7_latest.png) (D14 = UART1 TX) and capture the message? It should point to the source of the crash. Also, please attach your sketch.

guilleherreraf, 2021-02-24:
Here is the script @aentinger (it is pretty much the example):
[MCP2515-Loopbacktest.zip](https://github.com/107-systems/107-Arduino-MCP2515/files/6037540/MCP2515-Loopbacktest.zip)
Regarding the 3V3 TTL to USB converter I will try to find one and capture the message as you recommend.

aentinger, 2021-02-24:
Sketch looks okay :+1: we'll be needing the serial capture.

guilleherreraf, 2021-02-24:
I'll try to do that tomorrow and get back to you! Thanks a lot for your help.

guilleherreraf, 2021-02-25:
Hello @aentinger. I managed to find an USB to serial converter based on the CP2102N family. Now, when the error occurs, will it automatically output a message through the corresponding port? Or do I have to add something in the SW? I guess I should at least initialize the serial port with `Serial1.begin(baud_rate)`.

Workshopshed, 2021-08-22:
Had similar issues with E-Ink SPI device. Got the flashing warning light. Useful to know what that does.

PR #89 (2023-07-17): No longer list platform support for mbed and esp32.
This fixes #36.
This fixes #69.
This fixes #81.