107-systems/107-Arduino-MCP2515 #40: MKR Zero in combination with MKR Can Shield, sniffer won't loop

Timmytubby, 2021-04-08:
Dear developers,

i try to read out my Smart Fortwo 453. This would run on 500kbps can speed. I use the Can-Sniffer code from you at the moment, with only change in the can speed. When connected i receive the first ID and code, but after that i have to reset the board to receive a new message. I see that it is interrupt based, so there are several options to go bad in my case, why it isn't looping:
- Message stucture not correct?
- Buffer stays full, perhaps it should be cleared after serial printing.
- Something i overlooked....?

Hereby a message which i received from the car, but the first line is automatic, all the next lines need a arduino reset motivation.

20:39:01.075 -> [ 1288428] ID(RTR) 40000597 DATA[14] FE FE FE 0 8A FE FF F0 BC 0 0 20 BC 0 
20:39:24.069 -> [ 1872245] ID(RTR) 40000514 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:39:34.095 -> [ 2101521] ID(EXT)(RTR) D3489B6F DATA[1] FF 
20:42:18.324 -> [ 1920865] ID(EXT) 986F249A DATA[1] FF 
20:42:28.350 -> [ 1076353] ID(RTR) 40000597 DATA[14] FF 4D FE 0 8A FE FF F0 BC 0 0 20 BC 0 
20:42:31.334 -> [ 1153013] ID(EXT) 91489249 DATA[2] 49 4D 
20:42:34.351 -> [ 1464594] ID(EXT) 9149EF49 DATA[2] 49 4D 
20:42:37.359 -> [ 1894703] ID(EXT) 9649EF49 DATA[2] 49 4D 
20:42:40.356 -> [ 1765925] ID(EXT) 986EEF49 DATA[2] 49 4D 
20:42:44.360 -> [ 1728595] ID(RTR) 40000324 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:42:48.386 -> [ 1658366] ID A5 DATA[2] 49 24 
20:42:48.386 -> [ 1668403] ID(EXT) 91489249 DATA[2] 49 4D 
20:42:52.387 -> [ 1763048] ID A5 DATA[2] 49 24 
20:42:52.387 -> [ 1773072] ID(EXT) 91489249 DATA[2] 49 4D 
20:45:29.570 -> [ 1911046] ID(EXT) 91489249 DATA[2] 49 4D 
20:45:32.591 -> [ 1183547] ID(EXT) 986F249A DATA[2] 49 4D 
20:45:35.597 -> [ 1524641] ID(RTR) 40000597 DATA[14] 49 4D FE 0 8A FE FF F0 BC 0 0 20 BC 0 
20:45:39.627 -> [ 1816620] ID(EXT) 91489249 DATA[2] 49 4D 
20:45:42.608 -> [ 1361230] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:01.688 -> [ 1831922] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:07.711 -> [ 1859604] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:10.713 -> [ 1726421] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:13.738 -> [ 1879474] ID(RTR) 40000324 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:47:16.744 -> [ 1515905] ID(RTR) 40000324 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:47:19.725 -> [ 1834599] ID A5 DATA[2] 49 24 
20:47:19.763 -> [ 1844652] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:19.763 -> [ 1854460] ID(EXT) 9149EF49 DATA[2] 49 4D 
20:47:23.758 -> [ 1560153] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:28.772 -> [ 1440074] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:31.761 -> [ 1357774] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:34.786 -> [ 1566512] ID(RTR) 40000514 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:47:36.787 -> [ 921632] ID(EXT) 986F249A DATA[5] 24 AB FE 0 8A 
20:47:39.808 -> [ 1589707] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:42.798 -> [ 1664040] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:44.805 -> [ 1016505] ID(EXT) 91489249 DATA[2] 49 4D 
20:47:48.827 -> [ 1548796] ID(EXT) 9151A749 DATA[2] 49 4D 
20:47:51.816 -> [ 1605878] ID(EXT) 986EA749 DATA[2] 49 4D 
20:47:55.820 -> [ 1862204] ID(RTR) 40000324 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:47:58.827 -> [ 1420857] ID 1BF DATA[11] 49 4D FE 0 8A FE FF F0 BC 0 0 
20:48:19.873 -> [ 1130711] ID 1BF DATA[11] 49 4D FE 0 8A FE FF F0 BC 0 0 
20:48:22.841 -> [ 1042329] ID(RTR) 40000514 DATA[9] 49 4D FE 0 8A FE FF F0 BC 
20:48:25.888 -> [ 1730071] ID(RTR) 40000597 DATA[14] 24 AB FE 0 8A FE FF F0 BC 0 0 20 BC 0 
20:48:30.863 -> [ 1437531] ID(RTR) 40000597 DATA[14] 49 4D FE 0 8A FE FF F0 BC 0 0 20 BC 0 
20:48:33.861 -> [ 1721328] ID(RTR) 40000597 DATA[14] 49 4D FE 0 8A FE FF F0 BC 0 0 20 BC 0 
20:48:36.910 -> [ 1260549] ID(EXT) 9149E049 DATA[2] 49 4D 
20:48:38.880 -> [ 1121773] ID(EXT) 91489249 DATA[2] 49 4D 
20:48:42.919 -> [ 1440104] ID(EXT) 91489249 DATA[2] 49 4D 

Hopefully you can help me with this issue from me.

Regards,
TK

aentinger, 2021-04-16:
Definitely same issue as #38, consider implementing a ring buffer and moving serial output to `loop` as well as a higher serial baudrate.

closed 2021-04-16 by aentinger

Timmytubby, 2021-04-18:
Ok, thnx aetinger.

I tried already the first advice with falling to low interrupt detection. Didn't do the job.  Loop serial on higher baudrate, already done, no result.

Ring buffer: i will give it a go.

Regards,
TK