107-systems/107-Arduino-MCP2515 #47: is this compile error due to missing support?

DaleSchultz, 2021-07-24:
Hi

I am getting this error:

`sketch\ArduinoMCP2515.cpp.o:(.text._ZN7MCP251514MCP2515_Config13enableIntFlagENS_7CANINTEE[MCP2515::MCP2515_Config::enableIntFlag(MCP2515::CANINTE)]+0x0): undefined reference to MCP2515::MCP2515_Io::setBit(MCP2515::Register, unsigned char)`

Could this be because I am trying to build for ESP8266 which may not be supported? Or did I do something dumb?

aentinger, 2021-07-26:
Hi :wave:
ESP8266 is not supported - it may work but it's not guaranteed to work - which example do you attempt to build?

DaleSchultz, 2021-07-26:
Hi,

I was trying to build the sniffer.

Dale

aentinger, 2021-07-26:
I gave it a quick try with an outdated ESP8266 core on my system and can confirm that it does not work. Sorry, but ESP8266 is not supported by this library (it is also - intentionally - not listed in `library.properties`).

closed 2021-07-26 by aentinger