107-systems/107-Arduino-MCP2515 #48: Cannot connect to MCP215 with the Rasbpery Pi pico board

pepeRossRobotics, 2021-08-10:
Hello, I have been trying to use this library along with UAV CAN to make some proof of concepts for a project.
I have not been able to use this library along with the **ArduinoCore-mbed** (using the RP2040).
I just dont manage to get any communications via CAN even with the loopback example. The is no error message, just there is no communications.
My current wiring is:
```
static const byte MCP2515_SCK   = 18;
static const byte MCP2515_MOSI  = 19;
static const byte MCP2515_MISO  = 16;
static const byte MCP2515_CS    = 17;
static const byte MCP2515_INT   = 6;
```

I have been able to have both examples working with the standard **arduino-pico** library with the same wiring mentioned before, the issue is that it seems that that library is not compatible withe the UAVCAN library as when I try to compile the UAVCAN examples I get the following errors:
```
Error while detecting libraries included by /home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/libcanard/canard.c

Error while detecting libraries included by /home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/libcanard/canard_dsdl.c

Error while detecting libraries included by /home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/o1heap/o1heap.c

home/pepe/.arduino15/packages/rp2040/tools/pqt-gcc/1.3.1-a-7855b0c/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_632379/sketch/UAVCAN-Heartbeat-Publish.ino.cpp.o: in function `_ZN9LockGuardC4Ev':
/home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/utility/LockGuard.h:24: undefined reference to `crit_sec_enter'
/home/pepe/.arduino15/packages/rp2040/tools/pqt-gcc/1.3.1-a-7855b0c/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_632379/sketch/UAVCAN-Heartbeat-Publish.ino.cpp.o: in function `_ZN9LockGuardD4Ev':
/home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/utility/LockGuard.h:25: undefined reference to `crit_sec_leave'
/home/pepe/.arduino15/packages/rp2040/tools/pqt-gcc/1.3.1-a-7855b0c/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_632379/libraries/107-Arduino-UAVCAN/ArduinoUAVCAN.cpp.o: in function `_ZN9LockGuardC4Ev':
/home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/utility/LockGuard.h:24: undefined reference to `crit_sec_enter'
/home/pepe/.arduino15/packages/rp2040/tools/pqt-gcc/1.3.1-a-7855b0c/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_632379/libraries/107-Arduino-UAVCAN/ArduinoUAVCAN.cpp.o: in function `_ZN9LockGuardD4Ev':
/home/pepe/Arduino/libraries/107-Arduino-UAVCAN/src/utility/LockGuard.h:25: undefined reference to `crit_sec_leave'
collect2: error: ld returned 1 exit status

```

Any pointers would be greatly appreciated

aentinger, 2021-08-10:
Hi :wave: I'm currently on vacation. Even when not in vacation support requests are only answered once / week at my personal discretion.

aentinger, 2021-08-13:
As written over [here](https://github.com/107-systems/107-Arduino-UAVCAN/issues/121#issuecomment-898616759), [earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico) is not supported. You are free to add support for this code via PR though.

closed 2021-08-13 by aentinger