107-systems/107-Arduino-MCP2515 #53: Nano 33 BLE, Example locks up

chalkytoast, 2021-11-15:
### :bug: Bug Report

On a Nano 33 BLE with a MPC2515, the UAVCAN-Heartbeat-Publish example locks up. The problem appears to be related to locking and/or interrupts, as it hangs when trying to aquire a lock. I managed to get it work by commenting out the locking code and disabling the interrupt. 

I'm not entirely certain if this is truely a software bug. Perhaps this could also be caused by a wiring issue or misconfiguration? 

```C++
// CritSec-mbed.cpp
extern "C" void crit_sec_enter() {}
extern "C" void crit_sec_leave() {}

// UAVCAN-Heartbeat-Publish.ino in void setup()
pinMode(MKRCAN_MCP2515_INT_PIN, INPUT_PULLUP);
// attachInterrupt(digitalPinToInterrupt(MKRCAN_MCP2515_INT_PIN), onExternalEvent, FALLING);
```

**Minimum configuration to replicate the bug**
* Build Toolchain: 1.8.15
* Arduino Core: Mbed Nano 2.5.2
* Library Version: 107-Arduino-UAVCAN 1.4.0, 107-Arduino-MCP2515 1.3.2

aentinger, 2021-11-15:
I transferred the issue over to [107-Arduino-MCP2515](https://github.com/107-systems/107-Arduino-MCP2515) since the problems clearly start with the MCP2515 library.

aentinger, 2021-11-16:
Thank you for clarification :+1: Bit on clarification on my end, I do work for Arduino so I know the codebase so I don't see any issue with sending all those messages in the ISR or writing those couple of Serial messages from within the ISR :wink: I guess I'll have to wire up a board myself, but this will not happen soon-ish.

PR #68 (2022-09-12): Fix: No more missing MCP2515 events because the MCP2515 keeps the INTline low UNTIL all EVENTS have been handled.
This fixes #62 and fixes #53.
closed 2022-09-12 by aentinger