107-systems/107-Arduino-MCP2515 #57: Interrupt always low

HansEge, 2022-05-17:
Hi,

I've been debugging this for way too long and need help. I am able to send CAN frames out onto the CAN bus, however I am not able to receive frames. That is because the interrupt pin on the MCP2515 is constant low. The low state is triggered if I send a CAN frame or if there is any traffic on the bus. When the interrupt pin is in its low state the arduino stalls and i have to reset both the arduino and the MCP2515.

If I instead disconnect the interrupt pin from the arduino 33 BLE I can send multiple frames no problem. However the interrupt pin on the MCP2515 is still low which it shouldn't be. I am using the example code and have tried both the sniffer and loopback examples with giving me the same behavior.

I'm at a loss at how to proceed. Any suggestions are very much appreciated!

HansEge, 2022-05-17:
I've just tested on a MKR ZERO and everything seems to be working. Is the NANO 33 BLE not supported?

aentinger, 2022-05-17:
Sounds like this is related to #53.

Nano 33 BLE should work with this library, so its not per se unsupported, just has not been tested recently. As I'm unable to dedicate time to investigate this within the near future it may take a while to be investigated. If you find the issue I'll be glad to test your PR.

HansEge, 2022-05-17:
Thanks. 

I'll most likely move on to another platform. Maybe the MKR WIFI 1010. Have you tested the library on that device?

aentinger, 2022-05-18:
The `MKR WIFI 1010` software platform is [ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd), same as for Arduino Zero. Consequently you should be fine.

Closing as duplicate of #53.

closed 2022-05-18 by aentinger