107-systems/107-Arduino-MCP2515 #60: [Feature Request] Documentation on how to use "CanBitRate" enumeration

lordmundi, 2022-08-02:
[image]

### ⚡ Feature Request

**Please Describe The Problem To Be Solved**
I apologize if my google-fu is just failing me and this is documented somewhere, but I'm not sure what to select for the "CanBitRate" during initialization. For example, "BR_500kBPS_16MHZ"... I imagine that the 500kBPS is the can bus speed, but the "_16MHZ"... what does that refer to? I am wondering if this pertains to the oscillator that is used on the can bus interface chip. For example, my MCP2515 boards have an 8MHz oscillator on them. Should I change this enumeration to use the "_8MHZ" variants? Or is this frequency something else related to the canbus that I don't understand?

**(Optional): Suggest A Solution**
Update the example programs in "examples" directory to have a comment around the lines suggesting how users set the CanBitRate enumeration. Also the example in the README.md file.

aentinger, 2022-08-02:
The `_16MHz` or `_8MHz` suffix describe the speed of the crystal used to clock the MCP2515. The usual breakout boards come with either 8 or 16 MHz crystals and you have to select the right constant according to which . Care to create a PR with a comment explaining this?

lordmundi, 2022-08-02:
sure thing. just sent one.

aentinger, 2022-08-03:
Closing die to #61 .

closed 2022-08-03 by aentinger