107-systems/107-Arduino-MCP2515 #81: Arduino Nano 33 BLE, MCP2515-Loopback Example locks up

Reset19, 2023-03-16:
[image]

### :bug: Bug Report

Loopback code locks up with Arduino Nano 33 BLE to MCP2515 module.

SPI activity in the interrupt context causes platform to lock up. If I substitute my own interrupt handler and set a flag, I can call 
mcp2515.onExternalEventHandler() in the loop processing and all works as expected.  The driver code works when transactions are removed from the interrupt context. Adding any action that requests anything from the MCP during interrupt context locks up the platform. 

It is as if the SPI bus is inactive/clock disabled when an attempt to initiate a SPI transaction occurs during the interrupt.

The example loopback code produces the issue.

**Minimum configuration to replicate the bug**
Arduino IDE 2.0.4
Arduino Nano 33 BLE
Teyleten Robot MCP2515 CAN Bus Board Module TJA1050 Receiver SPI for Arduino.
TXS01108E Logic Level Module 8 Channel Bi-Directional High Speed Full Duplex 3.3V 5V module.
107-Arduino-MCP2515 (latest)

aentinger, 2023-03-20:
Hi @Reset19 ☕ 👋 Thank you for your report ;) I shall take a look at it. Performing SPI transactions within ISR context should be possible, but its come to my attention that all the many Arduino platforms do not exactly behave in the same way -.-

Reset19, 2023-03-20:
Thankyou! Let me know if I can try something.

aentinger, 2023-03-23:
Hi @Reset19 :coffee: :wave:

Can you please give #82 a spin and let me know of the results?

aentinger, 2023-03-23:
You can use this [link](https://github.com/107-systems/107-Arduino-MCP2515/archive/refs/heads/fix-mbed.zip) to download a zip file with the library version containing the changes :bow: .

Reset19, 2023-03-23:
Thankyou for the zip folder. I have removed the old library and put the new folder in its place.

It does not resolve the issue.

I have substituted an interrupt handler with just a read of the status (see below). If the "_transfer(instruction);" is included, it locks up. I am wondering if there is a security type protection in the ARM that is inhibiting access.

Let me know if there is something you would like me to try.

void onExternalEventHandlerJS(void)
{
  **rxStatus = mcp2515.readStatusJS();**
  rxPnd++;
}

uint8_t MCP2515_Io::statusJS()
{
  uint8_t const instruction = static_cast<uint8_t>(Instruction::READ_STATUS);

  _select();
  **//                       _transfer(instruction);**
  **//uint8_t const status = _transfer(0);**
  _deselect();

  //return status;
  return 0xEE;
}

Reset19, 2023-05-11:
None that I am aware of. I am using the same platform.

aentinger, 2023-05-12:
It's likely that the mbed cores don't love interrupted-based SPI transactions within an interrupt.

Any fixes I'll gladly test and merge, but I don't have the spare time for this right now.

PR #89 (2023-07-17): No longer list platform support for mbed and esp32.
This fixes #36.
This fixes #69.
This fixes #81.