107-systems/107-Arduino-MCP2515 #94: Allow Toggling The Use Of Only One TX Buffer

joerossrobotics, 2023-08-30:
[image]

### ⚡ Feature Request

**Please Describe The Problem To Be Solved**
Add a function to allow the user to toggle the use of only one TX buffer. Currently this happens if libcanard is included in order to prevent unintentional priority inversion while transmitting OpenCyphal/CAN frames. However, while implementing the [Kocherga Bootloader](https://github.com/Zubax/kocherga), I needed to work directly with CAN Frames, while also preventing these priority inversion issues.

**(Optional): Suggest A Solution**
I have added the function enableSingleTXBuffer(bool enable). See POC [Here](https://github.com/joerossrobotics/107-Arduino-MCP2515/tree/feature/single_tx_buffer).

aentinger, 2023-08-30:
This can be implemented better using a preprocessor define set within **CMakeLists.txt**. Let's finish #93 first and then tacke this issue.

aentinger, 2023-09-04:
Closing, see https://github.com/107-systems/107-Arduino-MCP2515/pull/93#issuecomment-1705571045 .