107-systems/107-Arduino-MCP2515 #96: [Bug Report]

sofresh007, 2023-09-12:
Hi im trying to get this running on my arduino r4 wifi, i installed the library and hooked up my mcp2515 as follows.

I changed the int pin to 2 in the code and cs pin to 10 in the code but still nothing shows up in the serial monitor. I also changed to 8mhz which i have on my chrystal.

  MCP2515 INT -----> DigitalPin 2 Arudino
  MCP2515 SCK -----> DigitalPin 13 Arudino
  MCP2515 SI  -----> DigitalPin 11 Arudino
  MCP2515 SO  -----> DigitalPin 12 Arudino
  MCP2515 CS -----> DigitalPin 10 Arudino
  MCP2515 GND -----> PowerPin GND Arudino
  MCP2515 VCC -----> PowerPin 5v Arudino

```C++
#include <SPI.h>
#include <107-Arduino-MCP2515.h>
/* ... */
static int const MKRCAN_MCP2515_CS_PIN  = 10;
static int const MKRCAN_MCP2515_INT_PIN = 2;
/* ... */
void onReceiveBufferFull(uint32_t const timestamp_us, uint32_t const id, uint8_t const * data, uint8_t const len)
{
  Serial.println(id, HEX);
}
void onTransmitBufferEmpty(ArduinoMCP2515 * this_ptr)
{
  /* You can use this callback to refill the transmit buffer via this_ptr->transmit(...) */
}
/* ... */
ArduinoMCP2515 mcp2515([](){ digitalWrite(MKRCAN_MCP2515_CS_PIN, LOW); },
                       [](){ digitalWrite(MKRCAN_MCP2515_CS_PIN, HIGH); },
                       [](uint8_t const d) -> uint8_t { return SPI.transfer(d); },
                       micros,
                       onReceiveBufferFull,
                       onTransmitBufferEmpty);
/* ... */
void setup()
{
  Serial.begin(9600);
  while(!Serial) { }

  SPI.begin();
  pinMode(MKRCAN_MCP2515_CS_PIN, OUTPUT);
  digitalWrite(MKRCAN_MCP2515_CS_PIN, HIGH);

  pinMode(MKRCAN_MCP2515_INT_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(MKRCAN_MCP2515_INT_PIN), [](){ mcp2515.onExternalEventHandler(); }, FALLING);

  mcp2515.begin();
  mcp2515.setBitRate(CanBitRate::BR_250kBPS_8MHZ); // CAN bitrate and clock speed of MCP2515
  mcp2515.setNormalMode();
}

void loop()
{
  uint8_t const data[8] = {0xDE, 0xAD, 0xBE, 0xEF, 0xDE, 0xAD, 0xBE, 0xEF};
  mcp2515.transmit(1 /* id */, data, 8 /* len */);
  delay(100);
}
```

is it something i've missed?

Best regards..

aentinger, 2023-09-12:
Why should something show up on the serial monitor? This sketch only transmits CAN data ... .

closed 2023-09-12 by aentinger

sofresh007, 2023-09-12:
sorry i took the wrong code, i meant the cansniffer. my bad